ArduPilot/apm_planner #1220: Error opening port: Permission denied

activesince93, 2020-08-31:
I am trying to run APM Planner 2 on Ubuntu 18.04. I have followed the exact instructions given in `ReadME.md` file. I was able to successfully install APM Planner 2.

Now, whenever I click on **Connect**, I get the following error. 

[image]

Here is the log for the same. 
```
[20200831 13:07:06.987 DEBUG] - APMToolBar: CONNECT SELECTED
[20200831 13:07:06.988 DEBUG] - APMToolBar: connectMAV  QObject(0x0) with sLink 1
[20200831 13:07:06.989 DEBUG] - SerialConnection::connect()
[20200831 13:07:06.989 ERROR] - serial connection: error  QSerialPort::SerialPortError(NoError)
[20200831 13:07:06.990 ERROR] - serial connection: error  QSerialPort::SerialPortError(PermissionError)
[20200831 13:07:06.990 ERROR] - Error opening port "Permission denied" trying again...
[20200831 13:07:06.991 DEBUG] - APMToolBar: CONNECT unselected
[20200831 13:07:07.992 DEBUG] - SerialConnection::connect()
[20200831 13:07:07.993 DEBUG] - SerialConnection::disconnect() QSerialPort(0x561962d339b0)
[20200831 13:07:07.993 ERROR] - serial connection: error  QSerialPort::SerialPortError(NotOpenError)
[20200831 13:07:07.994 DEBUG] - LinkManager::linkDisonnected:  "ttyS4" 1
[20200831 13:07:07.995 DEBUG] - APMToolBar setConnection: false
[20200831 13:07:07.996 DEBUG] - APMToolBar: updateLinkDisplay "ttyS4" : "115200"  connected: false
[20200831 13:07:07.997 ERROR] - serial connection: error  QSerialPort::SerialPortError(NoError)
[20200831 13:07:07.998 ERROR] - serial connection: error  QSerialPort::SerialPortError(PermissionError)
[20200831 13:07:07.999 ERROR] - Error opening port "Permission denied" trying again...
[20200831 13:07:08.000 DEBUG] - serial connection: object destroyed: QObject(0x561962d339b0)
[20200831 13:07:09.000 DEBUG] - SerialConnection::connect()
[20200831 13:07:09.001 DEBUG] - SerialConnection::disconnect() QSerialPort(0x561962407fd0)
[20200831 13:07:09.002 ERROR] - serial connection: error  QSerialPort::SerialPortError(NotOpenError)
[20200831 13:07:09.003 DEBUG] - LinkManager::linkDisonnected:  "ttyS4" 1
[20200831 13:07:09.004 DEBUG] - APMToolBar setConnection: false
[20200831 13:07:09.005 DEBUG] - APMToolBar: updateLinkDisplay "ttyS4" : "115200"  connected: false
[20200831 13:07:09.006 ERROR] - serial connection: error  QSerialPort::SerialPortError(NoError)
[20200831 13:07:09.007 ERROR] - serial connection: error  QSerialPort::SerialPortError(PermissionError)
[20200831 13:07:20.041 ERROR] - Error opening port "Permission denied"
[20200831 13:07:20.042 DEBUG] - serial connection: object destroyed: QObject(0x561962407fd0)
[20200831 13:07:20.043 DEBUG] - serial connection: object destroyed: QObject(0x561962c070d0)
```

billbonney, 2020-08-31:
You need to uninstall the modem manager. See doc link at top of read me

activesince93, 2020-08-31:
Followed the instructions [given here](https://ardupilot.org/planner2/docs/installation-for-linux.html#common-problems). Still same issue. 

[image]

closed 2020-09-01 by activesince93