ArduPilot/apm_planner #1225: Arducopter 4.0.5 tlog and dataflash issues:

AndKe, 2021-01-14:
DF log: https://drive.google.com/file/d/1SQfmSRZCV_ajVo9pVHE4t_sZ9PqmtBep/view?usp=sharing
This log will load, usually after APMPlanner2 becomes a little unresponsive and Linux asks to terminate it..

Errors reported are (many instances of): 
Logline 1084: Unknown data type: a when decoding ISBD
Logline 1084: No values within data message
Logline 1096: Unknown data type: a when decoding ISBD

.tlog : https://drive.google.com/file/d/1qI_MlQd_dtbGbb6pxZKwneggXsx6WJyo/view?usp=sharing

the tlog will fail to load with: 
[image]

Arne-W, 2021-01-17:
@AndKe I will have a look...

Arne-W, 2021-01-18:
The "new" ISBD data contains a format specifire (the "a") which is currently unknown by the parser. Unfortunately this messages breaks the "normal" logging as it contains 3 arrays of values which have to be interpreted using another message called ISBH.
First I can make a "fix / hack" which drops those messages while parsing the logs. The second step should be the correct handling but first I have to understand how this new thingy works....

AndKe, 2021-01-18:
Does this help ? : https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/LogStructure.h#L274

Arne-W, 2021-01-18:
I have found it already but it is to undertstand how the data is organized - there are some sequence numbers - and how they should /could be displayed. Nevertheless thanks for the hint.

Arne-W, 2021-01-19:
Ok the bin log parsing works now but the new data can not be visualized. Tlog is still failing I am working on it.

Arne-W, 2021-01-22:
@AndKe I found the problem. You tlog contains corrupted data. There is a [PARAM_REQUEST_READ](https://mavlink.io/en/messages/common.html#PARAM_REQUEST_READ) message from ID 252 which contains a wrong data field. The field name which should be named "param_id" is named  "text" which cannot be handled by the data model.
The easiest solution would be to suppress those messages as they are useless for plotting. But perhaps it is better to add a proper handling of messages containing such corruptions.
I your case it seems to be a software bug as the message is repeated several times and it always contains this error.

AndKe, 2021-01-25:
@Arne-W We had two GCS's that day, and a pymavlink script on the payload computer (that processed DATA32)

I did re-test the following configurations:
RPAS+pilot GCS
RPAS+Pilot GCS+ mavlink-payload
RPAS+Pilot GCS+ mavlink-payload + payload ground script
RPAS+Pilot GCS+ mavlink-payload + payload ground script + AP2 on payload computer

It's now clear that the issue is caused by the payload GCS.
The payload-GCS was a APMPlanner2 2.0.26 (the payload operator downloaded it as the most recent)
(Pilot GCS was 2.0.28-rc1 / master)

Unless something is changed since 2.0.26 - it may be related to the fact that AP2 is not always fetching parameters when started more than once. It seems somehow the autopilot may "get tired" of a restarting AP2, and stop responding to param requests.

Arne-W, 2021-01-25:
@AndKe Hmmm - I would say the problem is that both GCS use the same mavlinkID when sending requests, but I am not sure.
Nevertheless it seems that the second GCS (the old AP2) has an error in the request. The param_id field has the wrong name perhaps this leads to that strange behaviour because the autopilot does not understand the request?
Does this also happen if you use 2 GCS with 2.0.28 ??

AndKe, 2021-01-28:
@Arne-W Thank you. Sorry for the delay, today I tested with two GCS on the same firmware (current master)
The results confirm that this is an issue with running two AP2 - not limited to 2.0.26
Please see full description here: https://github.com/ArduPilot/apm_planner/issues/1228

commit b558146b (2021-01-30): Added support for parsing data with typecode 'a' like ISBD and ISBH data. Fixes #1225
commit be6acb19 (2021-01-30): Improved handling of corrupted data in tlogs. Fixes #1225
closed 2021-01-30 by Arne-W