ArduPilot/apm_planner #1226: Error Creating SSL context

efectn, 2021-01-20:
I'm trying upload firmware to my Ardupilot. But i'm getting "Error Creating SSL context". How can i solve it?

Arne-W, 2021-01-22:
Which plattform (linux mac windows),  which apm planner version, which flight controller hardware?

Use the latest version -> [https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1)

.... but I am sorry this version does not support firmware flashing - its broken

efectn, 2021-01-23:
> Which plattform (linux mac windows), which apm planner version, which flight controller hardware?
> 
> Use the latest version -> https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1
> 
> .... but I am sorry this version does not support firmware flashing - its broken

I tried on Arch Linux and Windows 10. I need hardware flashing. Which h version supports this feature? Also my card is APM Planner 2.6

Arne-W, 2021-02-08:
@efectn Sorry the new flashing page is still under development. If you able to build the AP2 from scratch you could have a look at this branch [https://github.com/Arne-W/apm_planner/tree/Custom_FW_2](https://github.com/Arne-W/apm_planner/tree/Custom_FW_2)
Downloading and flashing of FW already works - but there are still some cornercases that could be better....

Arne-W, 2021-05-18:
@efectn If you are still intrested - [Here](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.29-rc1) you can find the current pre release supporting firmware flashing again. The Appimage should work on most linux plattforms.

efectn, 2021-05-18:
I will test when i go to the school, thanks.

closed 2021-11-06 by efectn