ArduPilot/apm_planner #1228: Two AP2 GCS causes trouble for each other.

AndKe, 2021-01-28:
This is verified with current master. 2.0.28-rc1
Please see attached tlogs:
[2021-01-28 08-.zip](https://github.com/ArduPilot/apm_planner/files/5885594/2021-01-28.08-.zip)

The first tlog, (08:39) is created when the first AP2 GCS is started - referred to as "GCS1"
GCS1 loads all parameters just fine.
The second tlog, (08:40) is created when the first AP2 GCS is started - referred to as "GCS2"
GCS2 connects and start to load parameters.
GCS1 instantly starts to animate its parameter bar, with a count that is >100 behind. (so it does reload, but not by reading the same requests/responses)
GCS1 mavlink inspector discovers another MAVID 252 (GCS2?)
GCS2 mavlink inspector discovers another MAVID 252 (?)
none manage to fully load parameters. (give up/time out without message)
I end the test.

Arne-W, 2021-01-30:
@AndKe I wonder where the device with the MAVID 252 is created. As we have already seen in #1225 this device sends malformed mavlink messages...
Perhaps the easiest fix would be to allow the user to set the MAVID of the AP2 GCS?

AndKe, 2021-01-30:
the funny sysID 252 must be related to :
comm/MAVLinkProtocol.h:56:    constexpr static quint8 s_SystemID    = 252;
or 
QGC.h:64:const static int defaultSystemId = 252; // Using 252 to 'crudely' identify a log created by APM Planner 2

There is already some logic to warn against conflicting sysid in comm/MAVLinkProtocol.cc that could, maybe be used.

Adding a manual GCS sysid setting is a good idea in general, but any change to GCS ID will break/affect tlogreplay which is hardcoded.  (and should not be)
comm/TLogReplayLink.cc:217:                if (message.sysid == 255)

AndKe, 2021-01-30:
@Arne-W the funny sysID 252 must be related to :
comm/MAVLinkProtocol.h:56: constexpr static quint8 s_SystemID = 252;
or
QGC.h:64:const static int defaultSystemId = 252; // Using 252 to 'crudely' identify a log created by APM Planner 2

There is already some logic to warn against conflicting sysid in comm/MAVLinkProtocol.cc that could, maybe be used.

Adding a manual GCS sysid setting is a good idea in general, but any change to GCS ID will break/affect tlogreplay which is hardcoded. (and should not be)
comm/TLogReplayLink.cc:217: if (message.sysid == 255)

Arne-W, 2021-01-31:
@AndKe I made a fix and the ID should be configurable now. Would be cool if you could test it.
It could be interesting to see how the TLOG log look like if the rest is working.

AndKe, 2021-02-01:
Please see attached logs.
The GCS ID seems to work fine and is stored too.
It resolves some of the issues, like endless confusion about who should update/recieve paramaters.
I attribute much of the packet loss and communication issues to the fact that the usual SiK firmware is not meant for multipoint use.
[logs-2x-gcs.zip](https://github.com/ArduPilot/apm_planner/files/5903751/logs-2x-gcs.zip)

Thank you.

commit c780f15e (2021-02-05): Comms: Allow changing the Mavlink ID of APM Planner in advanced config dialog. Fixes #1228
closed 2021-02-05 by Arne-W