ArduPilot/apm_planner #1230: Make all parameter search async

wucke13, 2021-02-01:
It's very annoying if one has to wait for the search to yield results. Especially if the whole UI thread is blocked. That's a no-go.

Arne-W, 2021-02-03:
@wucke13 Which search do you mean? If you mean the loading bar on the left bottom - this is not a search bar it shows the progress of parameter downloading from the UAS.
We have to wait until all parameters are downloaded otherwise we can not populate several views.

wucke13, 2021-02-04:
Obviously I do not mean the bar at the bottom, but the text box with the default string "Filter" in the upper right area of the `Standard Params` & `Advanced Params` panel. 

This one:
[image]