ArduPilot/apm_planner #1231: Combine Standard Params and Advanced params panel

wucke13, 2021-02-04:
I find my self jumping between both all the time. I think the UX would be way better, if both were combined. Maybe some visual marker next to the `parameter name (parameter id)` that marks advanced parameters could be used instead to still communicate to the user which parameters require attention when changing.

Arne-W, 2021-02-09:
@wucke13 I thought about removing those 2 pages from the configuration because it is a lot of work to maintain them.
Surely you have seen that some parameters no longer work, which is due to the fact that some parameters have been renamed or removed.
Therefore, we have to adapt and maintain these pages for all ardupilot versions, which is relatively time-consuming.
Why don't you use the full parameter list? And which parameters are you using there?

wucke13, 2021-02-09:
> Therefore, we have to adapt and maintain these pages for all ardupilot versions, which is relatively time-consuming.

Not if you do it clever. It is possible to synthesize a reasonable good UI from the parameter definition files.

> Why don't you use the full parameter list?

For it to work, the search needs to Filter, e.g. reduce the number of rows in the table. Also the information is very impractical, because most of it is cut away. Having the info about the currently selected row in the param table in a separate table would be a good start.

Arne-W, 2021-02-21:
@wucke13 Doing things clever is always a good Idea :smile: but nevertheless it takes a lot of time to do such a change and I do not have the time to do such a restructuring at the moment - sorry.
Perhaps we can do something in the full parameter list to improve the user experience.