ArduPilot/apm_planner #1232: tlog is not writing until uncheck + check the box in settings

RoboSchmied, 2021-02-05:
The tlog file is not written until I uncheck + re-check the box "Enable Mavlink Logging (tlog)" in "APM Planner 2.0 Config" menu.

The checkbox is enabled when the program starts but no tlog file is created when connecting serial port.
 
Disabling the checkbox "Enable Mavlink Logging (tlog)" and enabling it again makes the app create the file and log to it.

I guess it is a problem while loading the settings.

branch: master
version: APM Planner v2.0.28-rc1      
platform: Linux 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

drobban, 2021-02-07:
Nice error report!

Would guess that the maintainer more then happily would accept a pull request if you are able to figure out the source of the problem.

Haven't worked with the source for a long while now, but more then willingly to help out if you have any questions.

Arne-W, 2021-02-14:
@RoboSchmied  For me it works :smile: I will have a closer look in the next days.

Arne-W, 2021-02-21:
@RoboSchmied  Ok I took a closer look and I am sorry  as I cannot reproduce your issue.
All flags that enable the logging itself are enabled by default. The settings are loaded while you can see the splash screen - long before anything can connect. In your home folder should be an apmplanner2 folder. Inside of this folder you can find some txt files with a name starting with apmlog_. Inside you should be able to find something like
``
ArduPilotMegaMAV APM Connected
[20210221 17:25:10.266 DEBUG] - LinkManger::startLogging() "/home/user/apmplanner2/tlogs/quadcopter//2021-02-21 17-25-10.tlog"
[20210221 17:25:10.267 DEBUG] - Start MAVLink logging "/home/user/apmplanner2/tlogs/quadcopter//2021-02-21 17-25-10.tlog"
``
If the file does not exist we have to look closer - cause it is not logged if opening fails.

RoboSchmied, 2021-02-23:
@Arne-W  you are right + thx.

This issue only occurs, when [INAV](https://github.com/iNavFlight/inav) is the source of the mavlink telemetry data. 
Using ardupilot as telemetry source makes the logs be written normally (no issue).

Inav (flight controller software) has a limited mavlink implemetation. 
[inav docs](https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md):
>MAVLink implementation in INAV is transmit-only.

Is this an INAV issue or should "APMPlanner" be changed to fix this issue?

Arne-W, 2021-03-04:
@RoboSchmied It looks like the logging starts only if an `ArduPilotMegaMAV` connects - thus connecting an INAV does not trigger logging start. It would be cool if you could upload a logfile where I can see what is happening when you are connecting to the INAV device. Perhaps we can do something....

RoboSchmied, 2021-03-15:
@Arne-W Thank you.
I uploaded an apmlog and a tlog file here: [tlog-file](https://www.datafilehost.com/d/f1fa0879)