ArduPilot/apm_planner #1235: Update build instructions to latest Ubuntu LTS

padcom, 2021-02-20:
The current set of instructions needed to build the app is tailored at an outdated version of Linux (I believe it is for 16.04 or even earlier). It means that both the compiled version as well as the build system won't work on the current LTS (20.04 at the time of creating this issue).

Please either provide appimage of the application (which would be very much preferred) or fix the instructions to build the app. Not everyone is familiar with QT, C++ and Python programming. Also, there is no other viable alternative for Linux as the mission planner is written in a version of .NET that is not fully compatible with mono.

I'd really appreciate having a version of mission planner that actually runs on Linux...

padcom, 2021-02-20:
A good way of testing if the build works as expected would be to create `Dockerfile` that uses the latest Ubuntu LTS as base and just run it all in the container. It will allow builds to be executed from any kind of operating system as Docker is supported on all major platforms.

That approach has been taken by iNav's firmware build system, and it works really well.

padcom, 2021-02-20:
I'd be willing to help create the `Dockerfile` if a viable set of commands, that work on Ubuntu 20.04 LTS would be provided.

Arne-W, 2021-02-21:
@padcom The latest release can be found [here](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.28-rc1) There you can find an appimage of AP2. Last year I made a script which builds AP2 in a 16.04 container an then creates a appimage. Unfortunately the building still works but the linuxdeploy tools for crating the appimage are crashing inside the container.
[Here](https://github.com/Arne-W/apm_planner/tree/Docker_%26_appimage) you can find the branch I used for building. In the docker/ubuntu-16.04-Qt5.12/ directory you will find the scripts.

padcom, 2021-02-22:
Merci! I see there's an appimage :) That's so cool! I don't know how I missed that. Maybe it'd make sense to add a link to releases to the official documentation? It seems to be linking only to old releases...

closed 2021-02-22 by padcom