ArduPilot/apm_planner #1239: PFD Altitude >1000m - display error.

AndKe, 2021-04-17:
[image]

The altitude tape ends, and both the first digit and the decimal is gone.
A quick fix could be to align the text so that the four first digits were visible (and the decimal gone)

Arne-W, 2021-05-16:
@AndKe will 2000m be enough?

AndKe, 2021-05-16:
@Arne-W  Yes, even for operations at high elevations, 2000 should be fine.  (I am surprised that you ask, because I would just think this is a formatting-logic issue and once fixed, anything to 9999 would work.)

Arne-W, 2021-05-16:
@AndKe Yes the formatting changed - removed the decimal when higher than 100m. But the scrolling numbers thingy (no Idea how to call them) ended at 1000m they will now end at 2000m.

AndKe, 2021-05-16:
Thank you for fixing the speed tape,  (also known as "scrolling numbers thingy")
:)

billbonney, 2021-05-16:
Apologies. I should have realized that people may fly over 120m (400ft) :-o

AndKe, 2021-05-16:
Yes, that may be less common, but even relatively standard commercial mapping flights of some rivers between a dam, and sea level, can require >1000m AMSL or relative altitude indication, while barely being over 100m AGL at most. (usually in the steepest parts)

commit a8bde8ea (2021-05-22): Allow 2000m in PFD altitude indicator. Fixes #1239
closed 2021-05-22 by Arne-W