ArduPilot/apm_planner #1240: Download for linux need update

AndKe, 2021-05-12:
This is where people that google for APM Planner 2 download end up:
https://ardupilot.org/planner2/docs/installation-for-linux.html#download
There's a link there leading to https://firmware.ardupilot.org/Tools/APMPlanner/
(which contains horribly outdated versions)

Today I discovered that appimage at https://github.com/ArduPilot/apm_planner/releases exists.

This appimagees should be added to the previously mentioned place, or the "installation" section should be updated to point to github*releases

Arne-W, 2021-05-15:
I don't have write access to https://firmware.ardupilot.org/Tools/APMPlanner/ therefore I can't upload the appimage there nor can I update the json file which contains the latest versions. I asked @billbonney to build a mac version and update those files but he seems to be very busy at the moment.

DoctorDizzee, 2023-03-07:
I am also having problems with installing the Linux version. I did notice the following errors when using The first command 
sudo apt-get update
sudo apt-get install qt5-qmake qt5-default.....

At the bottom will return:
Package qt5-default is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package qtdeclarative5-qtquick2-plugin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'qt5-default' has no installation candidate
E: Package 'qtdeclarative5-qtquick2-plugin' has no installation candidate

...

E: Unable to locate package python-serial
E: Package 'python-pexpect' has no installation candidate

Am I doing something wrong or are all the packages no longer available? 
[full-terminal.pdf](https://github.com/ArduPilot/apm_planner/files/10914181/full-terminal.pdf)

AndKe, 2023-03-08:
@DoctorDizzee that's outdated information.
install qt-community edition from https://www.qt.io/download-open-source
install 5.15.2 - and add it to you path (ulness it's done automatically.)
then you can skip installing any qt* packages using apt.

DoctorDizzee, 2023-03-09:
I tried with your instructions, still no luck.
I decided to move to QGroundControl and it works. I hope you get the Linux side working for the future. Thank you for your work, and quick response.

AndKe, 2023-03-09:
my environment for QGC is exactly the same.
are you sure you had the correct qmake?
```
$ qmake -v
QMake version 3.1
Using Qt version 5.15.2 in /home/andre/Qt/5.15.2/gcc_64/lib
``` 
Anyway: both QGC and AP2 are nice to have - QGC has nice and smooth planning GUI, but AP2 is much better at tuning and working with parameters.  Nothing beats AP2's fast and feature-rich log analysis and graphing.

DoctorDizzee, 2023-03-09:
I have given it another couple hours worth of effort and here is where I have landed:
  Package libssl1.1 is not installed.
This is no longer available for install on the apt store due to discontinuation.
[full-term.pdf](https://github.com/ArduPilot/apm_planner/files/10937041/full-term.pdf)

AndKe, 2023-03-10:
you are trying to install old version 2.0.26 used a .deb package.
If you built it from source - you would get 2.0.30 ... there is also a .Appimage for a newer build here: https://github.com/ArduPilot/apm_planner/releases/tag/2.0.30-rc2
you'll be better off with that.

DoctorDizzee, 2023-03-10:
That works for me! Thank you for your help!

DoctorDizzee, 2023-07-31:
I have moved to windows to run this program since, but thank you for updating it. Many Linux users out there will appreciate it.