ArduPilot/apm_planner #1248: Uploading/downloading mission times out

AndKe, 2021-07-02:
interestingly: uploading mission fails with timeout, but the process actually suceeds (the mission is actually uploaded)

[20210702 15:19:23.570 WARN ] - Finally timed out - going to idle. Current state was: 4
[20210702 15:19:27.568 WARN ] - Timeout requesting waypoint count - retrying.
[20210702 15:19:29.560 WARN ] - Timeout requesting waypoint count - retrying.
[20210702 15:19:31.564 WARN ] - Timeout requesting waypoint count - retrying.
[20210702 15:19:33.572 WARN ] - Timeout requesting waypoint count - retrying.

timtuxworth, 2022-10-18:
I've connected to an existing vehicle that has an existing mission on it. I can't download the mission from the vehicle. I get a timeout.

Arne-W, 2022-10-31:
@timtuxworth could you please tell some details? Linux or macOS? Version of Apm-Planner? It would also be cool if you could copy the content of the log.

timtuxworth, 2022-10-31:
Hi @Arne-W this is on MacOS, 2.0.30-r1. Happy to get you a log - from where? I'm assuming you don't mean the flight log.

Arne-W, 2022-11-01:
Hmm - I have no idea where you can find the log on macOS. On Linux there is a file called "apmlog_yyyymmdd.txt" in your "home/apmplanner2" directory. Perhaps its there on macOS too??

timtuxworth, 2022-11-01:
Yes - pretty much the same - under my "/users/tim/apmplanner2". Here's an example.

[apmlog_20221025.txt](https://github.com/ArduPilot/apm_planner/files/9912217/apmlog_20221025.txt)

Arne-W, 2022-11-01:
Cool - I will have a closer look in the next days. :+1:

Arne-W, 2022-11-10:
@timtuxworth Sorry but I am not able to reproduce it. Have you tried to refresh the list manually? On the bottom there should be a refresh button. 
The Implementation requests the next waypoint as soon as a waypoint is received. So if it times out I would say either the request or the answer is lost.... What sort of connection are you using?? I tested with USB.

AndKe, 2022-11-10:
@timtuxworth  is that by any chance a 3DR Solo with OpenSolo ? - I remember that there was some issues with the controller-scripts that were/are not fully MAVLink2 compatible.

AndKe, 2023-02-02:
This is fully reproducible with a serial connection to a ArduPilot, and SITL.
I also believe this issue https://github.com/ArduPilot/apm_planner/issues/1270 is related .. he points to a specific issue with more than one message sending events

AndKe, 2023-02-02:
FWIW: APM-Planner_2-2.0.30-rc1-x86_64.AppImage  works fine..  that's only 16 commits since..

What would be the procedure for me to build some of them in a "random" order to narrow down which is to blame?
(what are the git commands I need to do so?)

AndKe, 2023-02-03:
@timtuxworth @Arne-W 
I did review the commits since the official .appimage was made.
I start to suspect the issue is somehow related to the build environment.
It would be of great help to have a new .appimage build from current master.
- Who can build an appimage? the daily auto-build is broken.

Arne-W, 2023-02-03:
@AndKe I can do a new build this weekend - sorry but you know SOOOOOOO many things to do and so little time

AndKe, 2023-02-03:
absolutely .. is there some way I can help- please say .. I am a Linux user, but never fiddled with building on github. If it is trivial - like change in upload paths or something - might be of help...otherwise, I doubt it.
Autobuilds would certainly help many more people.

Arne-W, 2023-02-04:
Autobuild isn't that trivial because we need at least Linux and Macos. Windows would be nice too. Moreover there is the Qt incompatibility with OpenSSL 3.x which makes it complicated to build on Ubuntu 22.04.

I made a new release candidate and a new Appimage. You can find it [here](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.30-rc2).
Unfortunately I am not able to build a Mac release - so sorry.

AndKe, 2023-02-04:
@Arne-W  Thank you very much. The appimage works perfectly with both waypoints, and firmware-fetching - (where both fail on my local builds (ubuntu 22.10)

Please tell about Qt incompatibilty with OpenSSL 3.x   (which OpenSSL version is required/you use)
Also: which distro did you use to build it?

Arne-W, 2023-07-10:
@timtuxworth Is this issue still active? Did you try to build the newest version (2.30.rc3)?

timtuxworth, 2023-07-10:
Hi Arne I don't have a build environment for APM Planner. If there was a dmg I would try it.

Arne-W, 2023-07-11:
We will do a release including Mac in the next weeks - I will inform you.