ArduPilot/apm_planner #1251: Graph: issue with time between, and incorrect mode changes

AndKe, 2021-11-03:
Please see described log here: https://drive.google.com/file/d/1SOBGs9QMW2ZKAVyC0Fy1heO6wfvezv4T/view?usp=sharing

Loading this log (various text unchecked) shows actual activity if I select altitude (or other data that indicate use)
[image]

Zooming in, reveals repeating mode-changes (that are not real)
[image]

Otherwise, it works great. :)

AndKe, 2021-11-04:
@Arne-W  - I kind of feel as if you have commented on both issues before, but can't find such thread.

Arne-W, 2022-03-12:
@AndKe The problem are the GCS messages which were treated as vehicle messages. That was easy to fix. But this log has several issues with the timeline. There are a lot of jumps and as you can see the log duration it is more than 417881829,395 YEARS :smile:

closed 2022-03-12 by Arne-W

AndKe, 2022-03-12:
@Arne-W   Thank you !

commit fdc0557a (2022-03-12): Graph View: Ignore GCS messages when parsing Tlogs. Fixes #1251