ArduPilot/apm_planner #1253: Graph: please add "instance" handling

AndKe, 2022-02-03:
I am attaching a newer log.
https://drive.google.com/file/d/1hILWLgsautPNggQ1I7Cxu0whEbaCuhA8/view?usp=sharing

Please handle IMU, IMU2 ,IMU3  and other messages with the new "instance" attribute.
@Arne-W  - could you please look into this? - Graphing is something you know well.

Arne-W, 2022-02-12:
@AndKe I will have a look - don't know how long it will take but I will do my best!

AndKe, 2022-02-12:
@Arne-W  Much appreciated! logic-wise, it seems not too bad, this is the mail issue that messes with AP2 as a great log-analysis tool for newer logs.

Arne-W, 2022-02-12:
Had a closer look now - at moment I don't have a good Idea how to fix it. It breaks the generic structure of the log -> one format description for every measurement. In this case we have one format description for several similar measurements. At the moment I have no idea how to detect that the field is an index an not a usual measurement.

AndKe, 2022-02-12:
@Arne-W  please see the format messages related to IMU (other Multi-ID packets look similar.)

2022-02-03 14:44:20.42: FMT {Type : 162, Length : 54, Name : IMU, Format : QBffffffIIfBBHH, Columns : TimeUS,I,GyrX,GyrY,GyrZ,AccX,AccY,AccZ,EG,EA,T,GH,AH,GHz,AHz}
2022-02-03 14:42:47.08: FMTU {TimeUS : 20654076, FmtType : 162, UnitIds : s#EEEooo--O--zz, MultIds : F-000000-----00}

Arne-W, 2022-02-12:
@AndKe Yes I have seen that. BUT the unit data is handled in a generic way especially the unit id data.
At the moment we have a select able entry for every FMT type and its childs, now we will have multiple entrys for some FMT types depending on their UnitID data. The unit data is applied when fetching data from the storage but all IMU data, for example, is already stored in one data row....
What I am trying to say is - It makes everything less generic and more complicated. :disappointed:

AndKe, 2022-02-12:
"and more complicated"   ... @Arne-W but of course, that could have been the intention behind that change :)
- I hope you can think of a solution - it's way over my head. and even MAVProxy is not fully/nicely supporting it yet (missing autocomplete and does not reveal the range)

AndKe, 2022-02-12:
FYI: I was too quick to ask here : https://discuss.ardupilot.org/t/where-can-i-find-information-on-the-new-log-format-with-multiple-ids/81817/3

Arne-W, 2022-02-20:
@AndKe Seems like its done :smile: 
I still have to cleanup the commits - but if you want to test: [https://github.com/Arne-W/apm_planner/tree/feature/Instance_handling](https://github.com/Arne-W/apm_planner/tree/feature/Instance_handling)
Let me know how you like it!

AndKe, 2022-02-20:
@Arne-W  Thank you!
When I saw the notification email I dropped "everything" and compiled it right away!  - it's fantastic!
I like how you added a "tree-level" to the instances, that's a nice touch.

I have two feedback comments:
1: Instance handling calls for an increase in AP2 version number. (it's a significant feature, lack of which would reduce AP2's usability for newer logs)

2: if this area can be made wider by default, that would be nice (if it's a simple change)
[image]

Arne-W, 2022-02-27:
@AndKe If you like to test - Presets should work now and the size of "the area" is now persisted.
For increasing the number we a new splash screen as the number is part of it....

If you don't find an issue we can do a new release in the next days...

AndKe, 2022-02-27:
@Arne-W  I just tested it, and it works beautifully.  
I did not understand what you mean by "new splash screen"

Arne-W, 2022-02-27:
@AndKe This is the so called splash screen

[image]

And as you can see it says 2.0 - and I don't know the font...

AndKe, 2022-02-27:
@Arne-W   Ah! - fantastic  - yes, that kind of splash screen I certainly know :)   - I was thinking you added something related to graphing :)   - tried to look for one when loading a log.

It brings me right to another idea; is it trivial to open a log directly from the terminal? (so that ```apmplanner2 log40.bin``` starts AP2 and loads the log(s) ?)

AndKe, 2022-03-07:
Please pull into master/make a release

commit d52d5995 (2022-03-12): Graph View: Support indexed data types flash log files. Fixes #1253
closed 2022-03-12 by Arne-W