ArduPilot/apm_planner #1258: "Install Firmware" gives server error output

yasemindeki, 2022-03-22:
"Install Firmware" gives the following error output:
Started downloading http://firmware.ardupilot.org/Copter/stable/PX4-quad/ArduCopter-v2.px4
redirect to https://firmware.ardupilot.org/Copter/stable/PX4-quad/ArduCopter-v2.px4
Error downloading firmware.
Error Number: 203
Error Text: Error transferring https://firmware.ardupilot.org/Copter/stable/PX4-quad/ArduCopter-v2.px4 - server replied: Not Found

yasemindeki, 2022-03-22:
"PX4-quad" named folder does not exist in https://firmware.ardupilot.org/Copter/stable/ address.

Arne-W, 2022-03-22:
@yasemindeki Please use this version: [https://github.com/ArduPilot/apm_planner/releases/tag/2.0.29-rc1](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.29-rc1)

yasemindeki, 2022-03-23:
Thank you so much.

closed 2022-03-23 by yasemindeki