ArduPilot/apm_planner #1259: Firmware download fails due to SSL certificate

AndKe, 2022-04-27:
```
[20220427 10:31:05.985 DEBUG] - Address: "127.0.0.1"
[20220427 10:31:06.024 DEBUG] - TryCorrectGoogleVersions, VersionGoogleSatellite:  "923"
[20220427 10:31:06.081 ERROR] - ApmSoftwareConfig::apmParamNetworkReplyFinished() Unable to retrieve pdef.xml file! Error num: QNetworkReply::SslHandshakeFailedError : "SSL handshake failed"
[20220427 10:31:06.088 DEBUG] - SCW: Link is disconnected
[20220427 10:31:06.097 DEBUG] - Baud rate: 115200 Expected: 4
[20220427 10:31:08.698 DEBUG] - APMToolBar: INITIAL SETUP SELECTED
[20220427 10:31:08.698 DEBUG] - APMToolBar: clear selected buttons
[20220427 10:31:08.698 DEBUG] - APMToolBar: FLIGHT DATA unselected
[20220427 10:31:08.699 DEBUG] - APMToolBar: INITIAL SETUP unselected
[20220427 10:31:08.743 DEBUG] - ApmCustomFirmwareConfig: Install Firmware selected
[20220427 10:31:08.744 DEBUG] - ApmCustomFirmwareConfig: Fetching available firmwares from:  "https://firmware.ardupilot.org/manifest.json"
[20220427 10:31:08.752 DEBUG] - APMToolBar: INITIAL SETUP SELECTED
[20220427 10:31:09.010 WARN ] - QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[20220427 10:31:09.012 DEBUG] - ApmCustomFirmwareConfig: Firmware fetch Error:  QNetworkReply::SslHandshakeFailedError
[20220427 10:31:11.024 DEBUG] - "retrieve versionobject from server: https://firmware.ardupilot.org/Tools/APMPlanner/apm_planner_version.json"
[20220427 10:31:11.304 WARN ] - QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[20220427 10:31:11.315 DEBUG] - Downloading: 0 / 0
[20220427 10:31:11.317 DEBUG] - AutoUpdateCheck::httpFinished()

```

Arne-W, 2022-10-31:
@AndKe is this still an issue?? I never could reproduce it.

AndKe, 2022-10-31:
@Arne-W unfortunately - yes. This issue is still going strong for other Qt things too, here is the related QGC issue:
https://github.com/mavlink/qgroundcontrol/issues/10269

Except for APMPlanner and QGC , I've not seen any other app with this issue.

Arne-W, 2022-11-01:
I just tested ubuntu 20.04 with 2.0.30-r1 appimage and it works like a charm
```
[20221101 17:01:41.210 DEBUG] - ApmCustomFirmwareConfig: Install Firmware selected
[20221101 17:01:41.212 DEBUG] - ApmCustomFirmwareConfig: Fetching available firmwares from:  "https://firmware.ardupilot.org/manifest.json"
[20221101 17:01:41.241 DEBUG] - APMToolBar: INITIAL SETUP SELECTED
[20221101 17:01:44.219 DEBUG] - ApmCustomFirmwareConfig: Fetching available firmwares finished
[20221101 17:01:44.364 DEBUG] - ApmCustomFirmwareConfig: Firmware list version is  "1.0.0"
[20221101 17:01:44.364 DEBUG] - ApmCustomFirmwareConfig: Found  30921  firmwares.
[20221101 17:01:44.491 DEBUG] - Found  22455  elements ins category  "STABLE"
[20221101 17:01:44.499 DEBUG] - Found  97  elements for platform  "edge"
[20221101 17:01:44.499 DEBUG] - Found  1  elements for mavtype  "ANTENNA_TRACKER"
[20221101 17:01:44.499 DEBUG] - Found valid Firmware for version:  "1.1.0"  on index:  0
[20221101 17:02:00.597 DEBUG] - Found  344  elements for platform  "fmuv3"
[20221101 17:02:00.598 DEBUG] - Found  3  elements for mavtype  "ANTENNA_TRACKER"
[20221101 17:02:00.599 DEBUG] - Found valid Firmware for version:  "1.1.0"  on index:  70
[20221101 17:02:00.600 DEBUG] - Found valid Firmware for version:  "1.1.0"  on index:  71
[20221101 17:02:00.600 DEBUG] - Found valid Firmware for version:  "1.1.0"  on index:  72
[20221101 17:02:06.741 DEBUG] - Found  94  elements for mavtype  "HELICOPTER"
[20221101 17:02:06.743 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1128
[20221101 17:02:06.744 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1129
[20221101 17:02:06.746 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1130
[20221101 17:02:06.747 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1131
[20221101 17:02:21.340 DEBUG] - Found  91  elements for mavtype  "Copter"
[20221101 17:02:21.342 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1588
[20221101 17:02:21.343 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1589
[20221101 17:02:21.345 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1590
[20221101 17:02:21.346 DEBUG] - Found valid Firmware for version:  "4.3.0"  on index:  1591
[20221101 17:02:29.980 DEBUG] - Selected Firmware on index: 1588
[20221101 17:02:29.981 DEBUG] - With info: QJsonObject({"format":"hex","git-sha":"93448b71380c417644c9082b7b23e80fb982b626","latest":0,"mav-autopilot":"ARDUPILOTMEGA","mav-firmware-version":"4.3.0","mav-firmware-version-major":"4","mav-firmware-version-minor":"3","mav-firmware-version-patch":"0","mav-firmware-version-str":"V4.3.0","mav-firmware-version-type":"STABLE-4.3.0","mav-type":"Copter","platform":"fmuv3","url":"https://firmware.ardupilot.org/Copter/stable-4.3.0/fmuv3/arducopter_with_bl.hex","vehicletype":"Copter"})
[20221101 17:02:38.814 DEBUG] - ApmCustomFirmwareConfig::firmwareDownloadBtnClicked
[20221101 17:02:38.816 DEBUG] - Downloading  "https://firmware.ardupilot.org/Copter/stable-4.3.0/fmuv3/arducopter_with_bl.hex"
[20221101 17:02:40.158 DEBUG] - ApmCustomFirmwareConfig: downloading firmware finished successful
[20221101 17:02:40.215 DEBUG] - Suggested Filename:  "fmuv3_Copter_4.3.0.hex"

```
Do you have openssl installed?? If not please try if it helps.

Arne-W, 2022-11-01:
Hups - sorry didn't see the info about 22.04. I will try it with 22.04 in the next days....

Did you only test the appimage or have you build by yourself?? I found some hints it could be related to the appimage which does not contain the right ssl dependencies.

AndKe, 2022-11-01:
@Arne-W  last appimage i saw was horribly outdated - and I love the graphs you make possible. - so I build myself :)

Arne-W, 2022-11-02:
@AndKe All right - Finally I am able to reproduce it. I am using Kubuntu 22.04.

- Using the [appimage](https://github.com/ArduPilot/apm_planner/releases/download/2.0.30-rc1/APM-Planner_2-2.0.30-rc1-x86_64.AppImage) everything works!
- Bulting from scratch with the native Kubuntu Qt 5.15.3-1 (via apt-get) everything works!
- Bulting from scratch with Qt installed with the qt-unified-installer 5.15.2-0 it fails like you described.

It would be cool if you could test the appimage - perhaps it works for you too.
Nevertheless I will try to find a better solution.

AndKe, 2022-11-02:
@Arne-W 
You are right  - the appimage works as expected.
Do you believe this can be solved by simply using QT >= 5.15.3  ?

Arne-W, 2022-11-06:
@AndKe I am not sure - I could not find a 5.15.3 in the qt-unified-installer. It seems the 5.15.3 is only delivered with the Ubuntu 22.04?!
Newer version of Qt are 6.x.x which are somehow incompatible with old versions. I will check the effort to build the planner with with Qt6.x.x - maybe it will solve the problem.

AndKe, 2022-11-06:
@Arne-W  given how ...conservative... QGC is with 6.x ... I don't think its an easy task:
https://dev.qgroundcontrol.com/master/en/getting_started/index.html#native-builds

Arne-W, 2022-11-09:
Ok - the migration to Qt 6.x.x seems to be a lot of work - too much for the moment! 
So I think there are 2 solutions for this issue:

- Use the native Ubuntu/Kubuntu Qt version delivered with the OS when building from scratch. For me it was Qt 5.15.3-1 on Kubuntu 22.04
- Build an Appimage on an older Ubuntu version like 18.04 an use it on 22.04.

AndKe, 2022-11-09:
thank you for looking into this,
The "download" link at https://ardupilot.org/planner2/docs/installation-for-linux.html#installation-for-linux  points to very old builds.

please consider adding/replacing the link with https://github.com/ArduPilot/apm_planner/releases/

Arne-W, 2023-03-15:
@AndKe Just to conclude this issue. The problem is caused by Qt which is built against OpenSSL 1.1.1 due to export restrictions. Since Ubuntu 22.04 it uses openSSL 3.0.2. When building against a Qt version installed with the Maintenance tool the usage of SSL is broken.
When using the Qt version which can be installed via apt-get on Ubuntu 22.04 everything works as this Qt version is built against openSSL 3.0.2.

AndKe, 2023-03-15:
@Arne-W  - yes - it's basically the same issue (and solution?) as QGC has/will need https://github.com/mavlink/qgroundcontrol/issues/10269#issuecomment-1468078857

AndKe, 2023-07-31:
Solved by using the updated build information at:
https://github.com/ArduPilot/apm_planner#install-the-required-packages

closed 2023-07-31 by AndKe