ArduPilot/apm_planner #1262: problems with sik_uploader

humanoid2050, 2022-07-04:
I am using APM Planner v2.0.30-rc1, built from source and installed as a deb via debuild as described in the documentation.

After install, sik_upload components exhibit errors:

```
Flashing with firmware
./sik_uploader/firmware/2.0/radio~hm_trp.ihx
/usr/bin/python2: can't open file './sik_uploader/sik_uploader.py': [Errno 2] No such file or directory

Complete 2
```

As noted in the fix bug report #766, the python file `sik_uploader.py` is being installed to /usr/bin. This is incorrect because 
a) the file does not have executable permissions, so it can't actually be run as a binary
b) apm_planner doesn't actually bother to look in this location

Invoking the script manually results in the following error:
```
python /usr/bin/sik_uploader.py
  File "/usr/bin/sik_uploader.py", line 100
    if use_mavlink:
                  ^
TabError: inconsistent use of tabs and spaces in indentation
```
These errors were introduced by https://github.com/ArduPilot/apm_planner/commit/565ff6f23a46fcc983dbc2dac7a591c16349c299 in which all existing space indentation was converted to tabs, and the new code was written with spaces. Maybe @AndKe or @Arne-W know what happened there.

AndKe, 2022-07-05:
> These errors were introduced by 565ff6f

I've updated uploader for python 3.x , it's in a PR
https://github.com/ArduPilot/SiK/pull/72

Arne-W, 2022-07-05:
@humanoid2050 There is a Appimage for easy use. [You can find it here](https://github.com/ArduPilot/apm_planner/releases). 
[Direct Link](https://github.com/ArduPilot/apm_planner/releases/download/2.0.30-rc1/APM-Planner_2-2.0.30-rc1-x86_64.AppImage)

@AndKe take a look at #1263 - Will that fix this issue?

AndKe, 2022-07-05:
@Arne-W , yes I believe it will fix the issue, the problem is that most use Python 3 by default, and the old upload  script is not ok with that.

humanoid2050, 2022-07-05:
Thanks for the responsiveness. 

@AndKe I pulled the SiK repo and the python3 upload script worked fine with my 3DR radio.

@Arne-W The Appimage would work on my desktop where I originally found the issue, but I was intending to deploy the program directly on my robot, which uses aarch64. That's why I started building from source.