ArduPilot/apm_planner #1264: APM Planer 2.0 not connecting to FC but QgroundControl can / does.

JulianGoesPro, 2022-07-12:
I am primarily using my MacBook and moved from iNav to ArduPlane for a T1-Ranger VTOL build.

My FC (H743) connects flawlessly on iNav in the iNav configurator, connects to QGroundControl after flashing ArduPlane automatically but when "connecting" in AP Planer 2.0, nothing happens, the artificial horizon stays locked... I am not sure its actually connecting or even trying to connect since the GUI shows "connected" instantly when clicking on it.

[image]
[image]
[image]
[image]

Would be awesome if anyone could direct me to something to try or to look up to get to the bottom of this... so weird...

Thank you!

Arne-W, 2022-07-13:
@JulianGoesPro Please use [this version](https://github.com/ArduPilot/apm_planner/releases/download/2.0.29-rc1/apm_planner_2.0.29-rc1_osx.dmg) which can be found [here](https://github.com/ArduPilot/apm_planner/releases/tag/2.0.29-rc1)

JulianGoesPro, 2022-07-16:
@Arne-W Thank you very much... got a crash first but now it's connecting! Awesome!
[image]
[image]
[image]

Arne-W, 2022-10-31:
seems to be solved :smiley:

closed 2022-10-31 by Arne-W