ArduPilot/apm_planner #1265: No OSX dmg for latest version of apm_planner

andyp1per, 2022-09-17:
@Arne-W do you know why this is? I looked into building it but seem to be missing some key pieces from a directory called dev_src?

billbonney, 2022-09-18:
$HOME/dev_src is where I place all my project files on my Mac. 

The file you are missing is a ‘source’ dmg that can be the target to copy the resultant .app to. 

You can creat it easy enough using diskutility.

andyp1per, 2022-09-18:
> $HOME/dev_src is where I place all my project files on my Mac.
> 
> The file you are missing is a ‘source’ dmg that can be the target to copy the resultant .app to.
> 
> You can creat it easy enough using diskutility.

Bill, I was able to create a dmg successfully using ```macdeployqt -dmg apmplanner.app``` - is it possible that qt has rolled the disk image creation into the tool?

My other issue is the resultant dmg works on the mac I have built on, but not on a much older mac

billbonney, 2022-09-18:
They may have added the tool in. TBH, you really need the dmg anymore. I used it before to ship the FTDIUSBDrivers, but they are build in now. I would suggest you make like easy and just zip the apmplanner.app directory.

timtuxworth, 2022-10-18:
I've tried this on my newish Mac Pro M1 (ARM version) and also on my older Intel Core i5 Air and it works great on both. These are both running the latest OSX Monterey 12.6. I love the look and feel, its a very nice clear UI.

I do get a timeout loading waypoints from the controller.

andyp1per, 2022-10-19:
Just to close this off the right command for packaging is:
```
macdeployqt apmplanner.app -dmg -appstore-compliant
```
from the release directory. Seems to work well and I have uploaded the DMG here:

https://firmware.ardupilot.org/Tools/APMPlanner/apm_planner_2.0.30-rc1_osx.dmg

To get to build for older macs I had to:

- Use macos High Sierra (10.13.6)
- Install Qt 5.12 from qt.io (brew will not work)
- Install XCode 10.1

closed 2022-10-19 by andyp1per

Arne-W, 2022-10-31:
@andyp1per Thanks for the mac build!