ArduPilot/apm_planner #1280: Syncing APMPlanner with ArduPilot Firmware sever

Hwurzburg, 2023-03-09:
@Arne-W Glad to see some activity on this GCS
However, the ArduPilot firmware server is way out of date for versions of this
We need either to arrive at a procedure to keep the firmware server up to date or remove APMPlanner software from that server and point the wiki to the Github release page
Can you attend either the next DEVCALL or EURO DEVCALL to discuss?
you can also contact me or Tridge or Randy  on Discord 
thanks

billbonney, 2023-03-10:
The simplest solution would be to just put a link on the firemware.arupilot.org to the releases in github.

Move the old releases into the Archive folder of firmware.ardupilot.org. 

Then we don't need to upload release there anymore.

@Arne-W @AndKe Should comment if that is OK with them?

AndKe, 2023-03-10:
to me, it sounds like a great improvement over the current situation.

Arne-W, 2023-03-11:
For me it would be ok too. Creating a new release here on github is very simple therefore I would prefer this solution.

AndKe, 2023-03-11:
when I read "syncing" - I kind of assumed that would be some "on new release -> copy files to ap server" script that would handle it.

Arne-W, 2023-03-11:
I don't have write access to the Ardupilot Firmware server. Otherwise I could upload the the Appimage there too...
@AndKe There is a new release candidate. @billbonney It would be supercool if you could build a MacOS version of the rc.

Hwurzburg, 2023-03-11:
if you wish to detach APMPlanner from the ArduPilot system and not have the GCS software part of the firmware server that is fine, we have two other maintained and supported GCS systems...I suggest moving the wiki content over to the GITHUB repo wiki system for completeness like many other ArduPilot associated software apps do (like Yaapu telem and TX GCS)...otherwise, I believe implementing the scripts and discipline like MAVProxy and MissionPlanner do, to have the stable, beta(release candidates), latest(master) versions available like other ArduPilot software should not be difficult

Hwurzburg, 2023-03-11:
I have asked that this be discussed in the DevCall and EU Devcall so the devs can discuss.. hopefully you can attend one of the two

andyp1per, 2023-03-11:
We now have a macOS environment that can be used to build APM planner. It would be possible to add this to CI.

billbonney, 2023-03-11:
I am not sure how we got from "arrive at a procedure to keep the firmware server up to date or remove APMPlanner software from that server and point the wiki to the Github release page" to "if you wish to detach APMPlanner from the ArduPilot system"

I updated the wiki to point to the github.com releases  as  requested. I did mentioned we can put a link in the firmware.arupilot.org to link to github.com to get the latest releases. I don't see a need to have the downloads in two places. 

I also don't see any need to move the wiki. That would require work and it's only a small team.

It doesn't need to be complicated. I have messaged Tridge as my access to upload to firmware.ardupilot.org is broken. I then fix this issue. Thx

Hwurzburg, 2023-03-12:
my point is that we should have APMPlanner updating in the firmware server and NOT point to a git repo for software...and that it should be straightforward to do like other ArduPilot GCS and software groups....I am not in favor of pointing people to a git repo to get releases...

andyp1per, 2023-03-12:
I am favour of any improvement on the status quo which is essentially no new mac updates until I built one late last year. I don't have a problem where we point as long as its clear.

Arne-W, 2023-03-12:
@andyp1per Thanks for the information.
@Hwurzburg For me it would be best to participate the devcall on Monday - Wednesday its nearly impossible.

billbonney, 2023-03-12:
@andyp1per I'm happy to keep the CI working. If you can arrange access to mac instance you have. I was running the builds on a MacMini, but it's needs some TLC (overheating) but it's 13 yrs old! lolz

tridge, 2023-03-13:
@billbonney happy to give you access to my mac
is this the right key to allow in?
https://github.com/billbonney.keys

billbonney, 2023-03-13:
There are two keys there, i'd suggest using the first `ssh-rsa` key

andyp1per, 2023-03-13:
@billbonney just a word about the setup - I have updated all the components and installed Qt on tridge's mac. I have not updated macOS / XCode past High Sierra (I think) so that builds are reasonably backwards compatible with a range of macs.

Arne-W, 2023-03-15:
Just to inform all of you - in the Devcall on Monday we have decided that I will inform @tridge about every new release and he will upload the appimage to the ardupilot firmware server. We will also provide the appimages via the github release page.

AndKe, 2023-03-15:
Great! if anything, having releases updated there can also bring more attention to this GCS.