ArduPilot/apm_planner #1283: appimage crash on ubuntu 22.04

tridge, 2023-03-13:
tried new appimage APM-Planner_2-2.0.30-rc3-x86_64.AppImage on ubuntu 
console output attached
[startup.txt](https://github.com/ArduPilot/apm_planner/files/10962647/startup.txt)

billbonney, 2023-03-15:
Thanks @tridge I'll take a look. I just need to spool up a linux distro 👍

Arne-W, 2023-03-15:
Hmm - I am still surprised about it.
I can not reproduce it on Kubuntu 22.04 with X11 desktop. @tridge are you using wayland, a special X11 environment or X11 forwarding?
Nevertheless I will try some other configurations.

Arne-W, 2023-03-15:
Now I tested on a Ubuntu 22.04 and it works too. So at the moment I am not able to reproduce it.

billbonney, 2023-03-16:
Yep. I tried on Xubuntu 22.04 and it was working for me. Which I know is a developers most non useful response.

You startup end with 
```
[20230314 10:16:08.495 DEBUG] - QUrl("file:///tmp/.mount_APM-PlEatKN9/usr/bin/qml/PrimaryFlightDisplayQML.qml")
[20230314 10:16:08.520 WARN ] - The X11 connection broke: No error (code 0)
X connection to :0 broken (explicit kill or server shutdown).
```

Here's the same part of the log, but was successful for me 
```
[20230316 22:21:03.762 DEBUG] - MODE: 88
[20230316 22:21:03.762 DEBUG] - "D|VECTOR|STABILIZED"
[20230316 22:21:03.762 DEBUG] - MODE: 92
[20230316 22:21:03.762 DEBUG] - "D|AUTO|STABILIZED"
[20230316 22:21:03.762 DEBUG] - MODE: 66
[20230316 22:21:03.762 DEBUG] - "D|TEST"
[20230316 22:21:03.793 DEBUG] - QUrl("file:///tmp/.mount_APM-PlIFqYTw/usr/bin/qml/PrimaryFlightDisplayQML.qml")
[20230316 22:21:03.929 DEBUG] - QML Status: QQuickView::Ready
[20230316 22:21:03.930 DEBUG] - QUrl("file:///tmp/.mount_APM-PlIFqYTw/usr/bin/qml/PrimaryFlightDisplayQML.qml")
[20230316 22:21:03.946 DEBUG] - QML Status: QQuickView::Ready
[20230316 22:21:03.949 INFO ] - UASActionsWidget creating  UASActionsWidget(0x55905c232400)
[20230316 22:21:03.951 DEBUG] - QuickView: load settings
[20230316 22:21:03.951 DEBUG] - QuickView: load settings END
[20230316 22:21:03.971 DEBUG] - QuickView: save settings
[20230316 22:21:03.971 DEBUG] - QuickView: saving key:  "GCS GPS.GPS Fix ()"  type: text
[20230316 22:21:03.971 DEBUG] - QuickView: saving key:  "GCS GPS.GPS HDOP ()"  type: text
[20230316 22:21:03.971 DEBUG] - QuickView: saving key:  "GCS GPS.GPS Sats (n)"  type: text
```

Arne-W, 2023-12-15:
closing this as nobody was able to reproduce  it.

closed 2023-12-15 by Arne-W