ArduPilot/apm_planner #1285: Graph view: Flight modes displayed as numbers for quadplane

AndKe, 2023-06-19:
Flight modes are displayed as numbers, this may be related to newer ArduPilot releases or quadplane modes.
Traditional flight modes from the copter and plane logs work as expected.

the modes (QLOITER, QHOVER, CRUISE) are all found in uas/ArduPilotMegaMAV.h

A Qplane log: https://drive.google.com/file/d/1raW4YJSFi2nYCjZ728s9boFaHuqKqpNS/view?usp=sharing

@Arne-W  - is there a chance you can look at it?

AndKe, 2023-06-22:
@Arne-W  Thank you very much - in two days I am going away on a vacation without my laptop, so I can enjoy the results after 30 June.  Have a great summer.

closed 2023-07-14 by Arne-W

AndKe, 2023-07-15:
Hi, I am still on vacation, but I see your recent commits and all the good stuff you added. Thank you!

Arne-W, 2023-07-16:
@AndKe You are always welcome - If I have the time I do my best :smile:

AndKe, 2023-09-13:
@Arne-W  With a log from Arduplane 4.4.0 : I still see the numbers with AP2.0.30
Please try https://drive.google.com/file/d/1mW0g0orfJ7wGICsTKXrKYz8kJ28TZhLB/view?usp=drive_link

Arne-W, 2023-09-13:
@AndKe For me it works :smile: 
[image]

AndKe, 2023-09-13:
@Arne-W  - sorry to disturb you just to expose my failure to spot a git detached head situation.  :)

Arne-W, 2023-09-15:
No Problem :smiley:

PR #1288 (2023-07-14): Fix arduplane modes
Modes for fixed wing should be showed correctly in Graph view now. Also added support for event messages. Fixes #1285