ArduPilot/apm_planner #1286: Dmg for mac on the RC3

Iragne, 2023-06-23:
Hi all,

I'm currently using it for my personal project, i also update it to the rc3 by compiling the code and corrected the outdated framework. but it would be great to have an asset as part of the release of the rc3. is that possible ?

here the PR for mac 
https://github.com/ArduPilot/apm_planner/pull/1284

Arne-W, 2023-06-26:
Yes sure, it would be very cool to have someone who can create a Dmg for Mac. Are you interested in building this release only or would you like top participate in future builds too?
I would like to fix the #1285 issue and the problems with the version file and then release the official 2.0.30. I hope I will fix those issues in the next 4 weeks. If you could provide an universal binary for Mac then I will inform you when I push the new Tag!

billbonney, 2023-06-27:
I would suggest you drop building dmg and just create the APMPlanner2.app and then zip it. The dmg was only required when we were shipping the drivers that are no longer required.

This then just simplifies the build process

Iragne, 2023-07-07:
Hello all, I would be really interested to help the project and the community. Happy to have help in creating build for you and review some.

Iragne, 2023-07-13:
@billbonney let me know if this is working, 
https://github.com/Iragne/apm_planner/releases/tag/release-test

Iragne, 2023-07-13:
Additionally i created a script for everybody to be able to do the DMG as soon as you can build on mc 
https://github.com/ArduPilot/apm_planner/compare/master...Iragne:apm_planner:mac-dmg

skaman82, 2023-07-24:
> 

Unfortunately didn't worked for me. I could open the .dmg file and copy the app into applications folder. But every time I try to launch the apmplanner2 - I get an error saying the app is damaged
[image]

Arne-W, 2023-07-28:
@Iragne I created a new tag 2.0.30 it would be extremely cool if you could build a MAC release.

Iragne, 2023-08-14:
@skaman82 let me know if this new DMG is working for you ?

Arne-W, 2023-08-25:
@Iragne  Do you still want to build a MAC release for us? I am asking because I already tagged the repo and build an appimage for linux and therefore I would highly appreciate a Mac build from you as we could put them to the download server at the same time.

Iragne, 2023-08-25:
Hello @Arne-W , i already build it but I was waiting some addtionnal validation in that chat sorry if we get it worng
if you want the build it's that link that I updated 2 weeks ago
https://github.com/Iragne/apm_planner/releases/download/release-test/apmplanner2.dmg

Iragne, 2023-08-25:
I'm building now the tag you mention

Iragne, 2023-08-25:
here the link
https://github.com/Iragne/apm_planner/releases/download/release-test/apmplanner2-2.0.30.dmg 
@Arne-W

Arne-W, 2023-08-25:
@Iragne WOW - That was quick :+1:  but I can't  start the application contained within the dmg. Is it a universal binary or just an ARM image? I am not a Mac user just borrowed the mac of my Girlfriend for testing :smile:

Iragne, 2023-08-25:
Let me try to find the other m1 i have to check. maybe the fact that i'm building it on my mac is the issue.

Arne-W, 2023-08-25:
@Iragne Ok I now have access to 2 mac minis - one is a m1 and one is a x86. I tried to open the dmg on the m1 and this is the result:
 
[image]

Do you have any Idea?

Arne-W, 2023-08-25:
On the x86 Mac it looks like this:
[image]

Could it be possible that it is no universal binary and that the Qt framework is not embedded into the dmg package??

Iragne, 2023-08-28:
ok i finally make it work on an other mac. sorry but i had only one working 
Here the link, I will also create the tutorial because the Qt signin is a bit complex 
https://github.com/Iragne/apm_planner/releases/download/release-test/apmplanner2-2.0.30.app.zip 
@Arne-W

billbonney, 2023-08-28:
There's no need for the dmg. Since you have the app bundled. Delivery of it zipped is the most convenient form.

Iragne, 2023-08-29:
So it's here in my previous comment

Iragne, 2023-08-29:
here the PR that is ongoing https://github.com/ArduPilot/apm_planner/pull/1297

Arne-W, 2023-08-29:
@Iragne Very cool - thanks for your effort. I hope I will have some time tomorrow to test your builds!!

Arne-W, 2023-08-30:
@Iragne Take a look at your PR I had an Idea to fix the Universal binary issue - and very cool that you signed it!

Arne-W, 2023-08-30:
@Iragne Your build has no commit ID - do you have an idea why?
[image]

Iragne, 2023-08-31:
Hum let me check