ArduPilot/apm_planner #1289: Error building on macOS 13.4.1 (22F82)

valeriyvan, 2023-07-10:
```
apm_planner % make -j8             
...
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -O2 -std=gnu++1y  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -mmacosx-version-min=10.7 -Wall -Wextra -fPIC -D_TTY_NOWARN_ -DGIT_COMMIT=2.0.30-rc3-3-g6bd09e4f2 -DGIT_HASH=6bd09e4f268749c824daeaf995d7df7013fbf7a0 -DQT_NO_DEBUG -DQGC_USE_ARDUPILOTMEGA_MESSAGES -DNOMINMAX -DEXTERNAL_USE -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_QUICKWIDGETS_LIB -DQT_WIDGETS_LIB -DQT_QUICK_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_SQL_LIB -DQT_SERIALPORT_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"/Users/*****/apm_planner"' -I. -Ilibs/mavlink/include/mavlink/v2.0 -Ilibs/mavlink/include/mavlink/v2.0/ardupilotmega -Ilibs/alglib -Ilibs/opmapcontrol/src/core -Ilibs/opmapcontrol/src/internals -Ilibs/opmapcontrol/src/internals/projections -Ilibs/opmapcontrol/src/mapwidget -Ilibs/utils -Ilibs -Ilibs/opmapcontrol -Ilibs/thirdParty/quazip -I/opt/homebrew/Cellar/qt@5/5.15.10/src/3rdparty/zlib -Ilibs/lib/Frameworks/SDL2.framework/Headers -I. -Isrc -Isrc/ui -Isrc/ui/linechart -Isrc/ui/uas -Isrc/ui/map -Isrc/uas -Isrc/comm -Iinclude/ui -Isrc/input -Isrc/lib/qmapcontrol -Isrc/ui/mavlink -Isrc/ui/param -Isrc/ui/watchdog -Isrc/ui/map3D -Isrc/ui/mission -Isrc/ui/designer -Isrc/ui/configuration -Isrc/output -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtOpenGL.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtSvg.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtPrintSupport.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtQuickWidgets.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtWidgets.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtQuick.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtMultimedia.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtGui.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtQmlModels.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtQml.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtNetwork.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtXml.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtSql.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtSerialPort.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtTest.framework/Headers -I/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtCore.framework/Headers -Ibuild-release/moc -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/AGL.framework/Headers -Ibuild-release/ui -I/opt/homebrew/Cellar/qt@5/5.15.10/mkspecs/macx-clang -F/opt/homebrew/Cellar/qt@5/5.15.10/lib -o build-release/obj/CommConfigurationWindow.o src/ui/CommConfigurationWindow.cc
1 error generated.
make: *** [build-release/obj/main.o] Error 1
make: *** Waiting for unfinished jobs....
src/comm/serialconnection.cc:102:10: warning: comparison of different enumeration types in switch statement ('QSerialPort::SerialPortError' and 'QIODevice::OpenModeFlag') [-Wenum-compare-switch]
    case QSerialPort::NotOpen:
         ^~~~~~~~~~~~~~~~~~~~
In file included from src/comm/QGCFlightGearLink.cc:36:
In file included from src/ui/MainWindow.h:37:
build-release/ui/ui_MainWindow.h:13:10: fatal error: 'QtGui/QAction' file not found
#include <QtGui/QAction>
         ^~~~~~~~~~~~~~~
build-release/ui/ui_MainWindow.h:13:10: note: did not find header 'QAction' in framework 'QtGui' (loaded from '/opt/homebrew/Cellar/qt@5/5.15.10/lib')
1 error generated.
make: *** [build-release/obj/QGCCore.o] Error 1
In file included from src/uas/UAS.cc:20:
In file included from src/ui/MainWindow.h:37:
build-release/ui/ui_MainWindow.h:13:10: fatal error: 'QtGui/QAction' file not found
#include <QtGui/QAction>
         ^~~~~~~~~~~~~~~
build-release/ui/ui_MainWindow.h:13:10: note: did not find header 'QAction' in framework 'QtGui' (loaded from '/opt/homebrew/Cellar/qt@5/5.15.10/lib')
In file included from src/comm/QGCJSBSimLink.cc:35:
In file included from src/ui/MainWindow.h:37:
build-release/ui/ui_MainWindow.h:13:10: fatal error: 'QtGui/QAction' file not found
#include <QtGui/QAction>
         ^~~~~~~~~~~~~~~
build-release/ui/ui_MainWindow.h:13:10: note: did not find header 'QAction' in framework 'QtGui' (loaded from '/opt/homebrew/Cellar/qt@5/5.15.10/lib')
1 warning generated.
1 warning generated.
1 warning and 1 error generated.
In file included from src/ui/CommConfigurationWindow.cc:35:
In file included from src/ui/CommConfigurationWindow.h:40:
build-release/ui/ui_CommSettings.h:13:10: fatal error: 'QtGui/QAction' file not found
#include <QtGui/QAction>
         ^~~~~~~~~~~~~~~
build-release/ui/ui_CommSettings.h:13:10: note: did not find header 'QAction' in framework 'QtGui' (loaded from '/opt/homebrew/Cellar/qt@5/5.15.10/lib')
make: *** [build-release/obj/QGCFlightGearLink.o] Error 1
1 warning and 1 error generated.
make: *** [build-release/obj/UAS.o] Error 1
1 warning and 1 error generated.
make: *** [build-release/obj/QGCJSBSimLink.o] Error 1
1 error generated.
make: *** [build-release/obj/CommConfigurationWindow.o] Error 1
```

Iragne, 2023-07-10:
Let me check but look like a qt version issue

valeriyvan, 2023-07-12:
```
qmake -query QT_VERSION
5.15.10
```

Iragne, 2023-07-13:
@valeriyvan  can you post the result of the qmake -spec macx-clang apm_planner.pro

Iragne, 2023-07-13:
@valeriyvan  can you post the result of the qmake -spec macx-clang apm_planner.pro
Also can you try to remove and re-install your QT 
brew uninstall qt@5
brew install qt@5

I'm running the same version of QT compare to you now my only diff is my  MacOSX13.1.sdk compare to you  MacOSX13.3.sdk

valeriyvan, 2023-07-13:
> @valeriyvan can you post the result of the qmake -spec macx-clang apm_planner.pro

```
apm_planner %  qmake -spec macx-clang apm_planner.pro
Project MESSAGE: Qt version 5.15.10
Project MESSAGE: Mac build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /Users/*****/apm_planner DESTDIR /Users/*****/DronePro/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /Users/*****/DronePro/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
Project WARNING: Qt has only been tested with version 12 of the platform SDK, you're using 13.
Project WARNING: This is an unsupported configuration. You may experience build issues, and by using
Project WARNING: the 13.3 SDK you are opting in to new features that Qt has not been prepared for.
Project WARNING: Please downgrade the SDK you use to build your app to version 12, or configure
Project WARNING: with CONFIG+=sdk_no_version_check when running qmake to silence this warning.
```

valeriyvan, 2023-07-13:
> Also can you try to remove and re-install your QT
> brew uninstall qt@5
> brew install qt@5

Now error is different:

```
apm_planner % qmake -spec macx-clang apm_planner.pro   
Project MESSAGE: Qt version 6.5.1
Project MESSAGE: Mac build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /Users/*****/apm_planner DESTDIR /Users/*****/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /Users/*****/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
apm_planner % make -j8                                 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -O2 -std=gnu++1z  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -mmacosx-version-min=10.7 -Wall -Wextra -D_TTY_NOWARN_ -DGIT_COMMIT=2.0.30-rc3-3-g6bd09e4f2 -DGIT_HASH=6bd09e4f268749c824daeaf995d7df7013fbf7a0 -DQT_NO_DEBUG -DQGC_USE_ARDUPILOTMEGA_MESSAGES -DNOMINMAX -DEXTERNAL_USE -DQT_NO_DEBUG -DQT_QUICKWIDGETS_LIB -DQT_QUICK_LIB -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_SQL_LIB -DQT_SERIALPORT_LIB -DQT_QMLINTEGRATION_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"/Users/*****/apm_planner"' -I. -Ilibs/mavlink/include/mavlink/v2.0 -Ilibs/mavlink/include/mavlink/v2.0/ardupilotmega -Ilibs/alglib -Ilibs/opmapcontrol/src/core -Ilibs/opmapcontrol/src/internals -Ilibs/opmapcontrol/src/internals/projections -Ilibs/opmapcontrol/src/mapwidget -Ilibs/utils -Ilibs -Ilibs/opmapcontrol -Ilibs/thirdParty/quazip -I/opt/homebrew/src/3rdparty/zlib -Ilibs/lib/Frameworks/SDL2.framework/Headers -I. -Isrc -Isrc/ui -Isrc/ui/linechart -Isrc/ui/uas -Isrc/ui/map -Isrc/uas -Isrc/comm -Iinclude/ui -Isrc/input -Isrc/lib/qmapcontrol -Isrc/ui/mavlink -Isrc/ui/param -Isrc/ui/watchdog -Isrc/ui/map3D -Isrc/ui/mission -Isrc/ui/designer -Isrc/ui/configuration -Isrc/output -I/opt/homebrew/lib/QtQuickWidgets.framework/Headers -I/opt/homebrew/lib/QtQuick.framework/Headers -I/opt/homebrew/lib/QtOpenGL.framework/Headers -I/opt/homebrew/lib/QtSvg.framework/Headers -I/opt/homebrew/lib/QtPrintSupport.framework/Headers -I/opt/homebrew/lib/QtWidgets.framework/Headers -I/opt/homebrew/lib/QtMultimedia.framework/Headers -I/opt/homebrew/lib/QtGui.framework/Headers -I/opt/homebrew/lib/QtQmlModels.framework/Headers -I/opt/homebrew/lib/QtQml.framework/Headers -I/opt/homebrew/lib/QtNetwork.framework/Headers -I/opt/homebrew/lib/QtXml.framework/Headers -I/opt/homebrew/lib/QtSql.framework/Headers -I/opt/homebrew/lib/QtSerialPort.framework/Headers -I/opt/homebrew/include -I/opt/homebrew/include/QtQmlIntegration -I/opt/homebrew/lib/QtTest.framework/Headers -I/opt/homebrew/lib/QtCore.framework/Headers -Ibuild-release/moc -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/AGL.framework/Headers -Ibuild-release/ui -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I/opt/homebrew/share/qt/mkspecs/macx-clang -F/opt/homebrew/lib -o build-release/obj/main.o src/main.cc
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -O2 -std=gnu++1z  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -mmacosx-version-min=10.7 -Wall -Wextra -D_TTY_NOWARN_ -DGIT_COMMIT=2.0.30-rc3-3-g6bd09e4f2 -DGIT_HASH=6bd09e4f268749c824daeaf995d7df7013fbf7a0 -DQT_NO_DEBUG -DQGC_USE_ARDUPILOTMEGA_MESSAGES -DNOMINMAX -DEXTERNAL_USE -DQT_NO_DEBUG -DQT_QUICKWIDGETS_LIB -DQT_QUICK_LIB -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_XML_LIB -DQT_SQL_LIB -DQT_SERIALPORT_LIB -DQT_QMLINTEGRATION_LIB -DQT_TESTLIB_LIB -DQT_CORE_LIB -DQT_TESTCASE_BUILDDIR='"/Users/*****/apm_planner"' -I. -Ilibs/mavlink/include/mavlink/v2.0 -Ilibs/mavlink/include/mavlink/v2.0/ardupilotmega -Ilibs/alglib -Ilibs/opmapcontrol/src/core -Ilibs/opmapcontrol/src/internals -Ilibs/opmapcontrol/src/internals/projections -Ilibs/opmapcontrol/src/mapwidget -Ilibs/utils -Ilibs -Ilibs/opmapcontrol -Ilibs/thirdParty/quazip -I/opt/homebrew/src/3rdparty/zlib -Ilibs/lib/Frameworks/SDL2.framework/Headers -I. -Isrc -Isrc/ui -Isrc/ui/linechart -Isrc/ui/uas -Isrc/ui/map -Isrc/uas -Isrc/comm -Iinclude/ui -Isrc/input -Isrc/lib/qmapcontrol -Isrc/ui/mavlink -Isrc/ui/param -Isrc/ui/watchdog -Isrc/ui/map3D -Isrc/ui/mission -Isrc/ui/designer -Isrc/ui/configuration -Isrc/output -I/opt/homeb [... cut ...]
make: *** Waiting for unfinished jobs....
make: *** [build-release/obj/SerialConfigurationWindow.o] Error 1
make: *** [build-release/obj/main.o] Error 1
20 errors generated.
make: *** [build-release/obj/QGCCore.o] Error 1
20 errors generated.
make: *** [build-release/obj/QGCFlightGearLink.o] Error 1
20 errors generated.
make: *** [build-release/obj/CommConfigurationWindow.o] Error 1
1 warning and 20 errors generated.
make: *** [build-release/obj/UAS.o] Error 1
20 errors generated.
make: *** [build-release/obj/MainWindow.o] Error 1
```

valeriyvan, 2023-07-14:
> ok look like you did not reinstall Qt5 but Qt6

could sequence of following reinstall qt6 instead of qt5?
```
brew uninstall qt@5
brew install qt@5
```

valeriyvan, 2023-07-14:
It looks like build instructions and/or scripts break if there's more that one Qt version installed.
I uninstalled Qt completely, installed Qt@5.
Now I have new error:

```
apm_planner % qmake -spec macx-clang apm_planner.pro   
zsh: command not found: qmake
apm_planner % /opt/homebrew/opt/qt@5/bin/qmake -spec macx-clang apm_planner.pro
Project MESSAGE: Qt version 5.15.10
Project MESSAGE: Mac build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /Users/*****/apm_planner DESTDIR /Users/*****/apm_planner/release TARGET apmplanner2
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /Users/*****/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Magellan 3DxWare (manual override from command line)
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Camera View
Project WARNING: Qt has only been tested with version 12 of the platform SDK, you're using 13.
Project WARNING: This is an unsupported configuration. You may experience build issues, and by using
Project WARNING: the 13.3 SDK you are opting in to new features that Qt has not been prepared for.
Project WARNING: Please downgrade the SDK you use to build your app to version 12, or configure
Project WARNING: with CONFIG+=sdk_no_version_check when running qmake to silence this warning.
apm_planner % make -j8
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic libs/opmapcontrol/src/mapwidget/mapripform.ui -o build-release/ui/ui_mapripform.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/MainWindow.ui -o build-release/ui/ui_MainWindow.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/CommSettings.ui -o build-release/ui/ui_CommSettings.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/SerialSettings.ui -o build-release/ui/ui_SerialSettings.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/UASControl.ui -o build-release/ui/ui_UASControl.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/UASList.ui -o build-release/ui/ui_UASList.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/UASInfo.ui -o build-release/ui/ui_UASInfo.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/UASView.ui -o build-release/ui/ui_UASView.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/ParameterInterface.ui -o build-release/ui/ui_ParameterInterface.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/WaypointList.ui -o build-release/ui/ui_WaypointList.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/ObjectDetectionView.ui -o build-release/ui/ui_ObjectDetectionView.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/JoystickWidget.ui -o build-release/ui/ui_JoystickWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/HDDisplay.ui -o build-release/ui/ui_HDDisplay.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/MAVLinkSettingsWidget.ui -o build-release/ui/ui_MAVLinkSettingsWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/AudioOutputWidget.ui -o build-release/ui/ui_AudioOutputWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCSensorSettingsWidget.ui -o build-release/ui/ui_QGCSensorSettingsWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/watchdog/WatchdogControl.ui -o build-release/ui/ui_WatchdogControl.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/watchdog/WatchdogProcessView.ui -o build-release/ui/ui_WatchdogProcessView.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/watchdog/WatchdogView.ui -o build-release/ui/ui_WatchdogView.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/SlugsDataSensorView.ui -o build-release/ui/ui_SlugsDataSensorView.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/SlugsHilSim.ui -o build-release/ui/ui_SlugsHilSim.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/SlugsPadCameraControl.ui -o build-release/ui/ui_SlugsPadCameraControl.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/uas/QGCUnconnectedInfoWidget.ui -o build-release/ui/ui_QGCUnconnectedInfoWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/designer/QGCToolWidget.ui -o build-release/ui/ui_QGCToolWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/designer/QGCParamSlider.ui -o build-release/ui/ui_QGCParamSlider.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/designer/QGCActionButton.ui -o build-release/ui/ui_QGCActionButton.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/designer/QGCCommandButton.ui -o build-release/ui/ui_QGCCommandButton.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCMAVLinkLogPlayer.ui -o build-release/ui/ui_QGCMAVLinkLogPlayer.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCWaypointListMulti.ui -o build-release/ui/ui_QGCWaypointListMulti.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCUDPLinkConfiguration.ui -o build-release/ui/ui_QGCUDPLinkConfiguration.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCUDPClientLinkConfiguration.ui -o build-release/ui/ui_QGCUDPClientLinkConfiguration.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCTCPLinkConfiguration.ui -o build-release/ui/ui_QGCTCPLinkConfiguration.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/QGCSettingsWidget.ui -o build-release/ui/ui_QGCSettingsWidget.h
/opt/homebrew/Cellar/qt@5/5.15.10/bin/uic src/ui/map/QGCMapTool.ui -o build-release/ui/ui_QGCMapTool.h
[... cut ...]
make: *** [build-release/obj/QGCJSBSimLink.o] Error 1
In file included from src/uas/UAS.cc:20:
In file included from src/ui/MainWindow.h:45:
In file included from src/ui/JoystickWidget.h:37:
In file included from src/input/JoystickInput.h:40:
libs/lib/Frameworks/SDL2.framework/Headers/SDL.h:32:10: fatal error: 'SDL2/SDL_main.h' file not found
#include <SDL2/SDL_main.h>
         ^~~~~~~~~~~~~~~~~
1 warning generated.
1 warning and 1 error generated.
make: *** [build-release/obj/UAS.o] Error 1
```

Iragne, 2023-07-14:
Perfect now it's the issue associate to this PR, https://github.com/ArduPilot/apm_planner/pull/1291
You can take the PR and it will normally compile correctly.

Let me know if it's working, i'm waiting @Arne-W  to validate the PR but if you can also validate it on your side would be perfect

closed 2023-07-14 by Arne-W

Iragne, 2023-07-14:
@valeriyvan Now you should be able to use the master and it should compile

commit 6fcb63a1 (2023-07-14): Build: Update SDL2 and command line to compile. Fixes #1289 #1290