ArduPilot/apm_planner #1290: MacOS 13.1 build issue (SDL2 related)

daanaerts, 2023-07-10:
Hi, I'm having issues building the Current master build on macOS 13.1 (arm64/M1)

Followed the instructions
- clone 
- `qmake -spec macx-clang apm_planner.pro`
- `make -j8` <-- gives issue:

```
#include <SDL2/SDL_main.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make: *** [build-release/obj/main.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from src/comm/QGCFlightGearLink.cc:36:
In file included from src/ui/MainWindow.h:45:
In file included from src/ui/JoystickWidget.h:37:
In file included from src/input/JoystickInput.h:40:
libs/lib/Frameworks/SDL2.framework/Headers/SDL.h:32:10: fatal error: 'SDL2/SDL_main.h' file not found
#include <SDL2/SDL_main.h>
         ^~~~~~~~~~~~~~~~~
src/comm/serialconnection.cc:102:10: warning: comparison of different enumeration types in switch statement ('QSerialPort::SerialPortError' and 'QIODevice::OpenModeFlag') [-Wenum-compare-switch]
    case QSerialPort::NotOpen:
         ^~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [build-release/obj/QGCCore.o] Error 1
In file included from src/comm/QGCJSBSimLink.cc:35:
In file included from src/ui/MainWindow.h:45:
In file included from src/ui/JoystickWidget.h:37:
In file included from src/input/JoystickInput.h:40:
libs/lib/Frameworks/SDL2.framework/Headers/SDL.h:32:10: fatal error: 'SDL2/SDL_main.h' file not found
#include <SDL2/SDL_main.h>
         ^~~~~~~~~~~~~~~~~
In file included from src/uas/UAS.cc:20:
In file included from src/ui/MainWindow.h:45:
In file included from src/ui/JoystickWidget.h:37:
In file included from src/input/JoystickInput.h:40:
libs/lib/Frameworks/SDL2.framework/Headers/SDL.h:32:10: fatal error: 'SDL2/SDL_main.h' file not found
#include <SDL2/SDL_main.h>
         ^~~~~~~~~~~~~~~~~
1 warning generated.
1 error generated.
make: *** [build-release/obj/QGCFlightGearLink.o] Error 1
1 error generated.
make: *** [build-release/obj/UAS.o] Error 1
1 error generated.
make: *** [build-release/obj/QGCJSBSimLink.o] Error 1
In file included from src/ui/CommConfigurationWindow.cc:54:
In file included from src/ui/MainWindow.h:45:
In file included from src/ui/JoystickWidget.h:37:
In file included from src/input/JoystickInput.h:40:
libs/lib/Frameworks/SDL2.framework/Headers/SDL.h:32:10: fatal error: 'SDL2/SDL_main.h' file not found
```

@Arne-W  I've seen your activity on fixing macOS builds. Thanks for your work!

Arne-W, 2023-07-10:
It was not me who did that SDL fixing - it was @Iragne.
@Iragne Could you please also have a look at this issue?

Arne-W, 2023-07-14:
Hmm - this issue should have been closed with #1291....

closed 2023-07-14 by Arne-W