ArduPilot/apm_planner #1292: Freshly build app crashes an launch on macOS 13.4.1 (22F770820d) on Apple M1

valeriyvan, 2023-07-14:
In seems SDL2's code signature invalid.
Might be connected to #1291.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               apmplanner2 [50571]
Path:                  /Applications/apmplanner2.app/Contents/MacOS/apmplanner2
Identifier:            org.ardupilot.apmplanner2
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [50555]
Responsible:           Terminal [630]
User ID:               501

Date/Time:             2023-07-14 16:58:48.1122 +0300
OS Version:            macOS 13.4.1 (22F770820d)
Report Version:        12
Anonymous UUID:        2876CEAF-790C-150B-529E-F26E28705789

Sleep/Wake UUID:       B4C30901-1BBC-4590-8224-374CD4690286

Time Awake Since Boot: 36000 seconds
Time Since Wake:       5031 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @executable_path/../Frameworks/SDL2.framework/Versions/A/SDL2
Referenced from: <EF9FFD08-5C39-319E-A14F-CB9BB9436BD0> /Applications/apmplanner2.app/Contents/MacOS/apmplanner2
Reason: tried: '/Applications/apmplanner2.app/Contents/Frameworks/SDL2.framework/Versions/A/SDL2' (code signature invalid in <5A2A6C17-8E90-3F32-B652-0182EF06171D> '/Applications/apmplanner2.app/Contents/Frameworks/SDL2.framework/Versions/A/SDL2' (errno=1) sliceOffset=0x001A0000, codeBlobOffset=0x00161470, codeBlobSize=0x000090B0), '/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file), '/System/Library/Frameworks/SDL2.framework/Versions/A/SDL2' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x1a6cc9118 __abort_with_payload + 8
1   dyld                          	       0x1a6cd4d7c abort_with_payload_wrapper_internal + 104
2   dyld                          	       0x1a6cd4db0 abort_with_payload + 16
3   dyld                          	       0x1a6c608a8 dyld4::halt(char const*) + 328
4   dyld                          	       0x1a6c5d590 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4184
5   dyld                          	       0x1a6c5be18 start + 1964

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x000000016d292460   x3: 0x000000000000008b
    x4: 0x000000016d292060   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000b80
    x8: 0x0000000000000020   x9: 0x0000000000000009  x10: 0x0000000000000001  x11: 0x000000000000000a
   x12: 0x0000000000000000  x13: 0x0000000000000035  x14: 0x0000000000000004  x15: 0x0000000000008000
   x16: 0x0000000000000209  x17: 0x00000001a6c5935c  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016d292060  x21: 0x000000000000008b  x22: 0x000000016d292460  x23: 0x0000000000000001
   x24: 0x0000000000000006  x25: 0x000000020247d100  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x000000016d292030   lr: 0x00000001a6cd4d7c
    sp: 0x000000016d291ff0   pc: 0x00000001a6cc9118 cpsr: 0x00001000
   far: 0x0000000103974000  esr: 0x56000080  Address size fault

Binary Images:
       0x102b6c000 -        0x1035bbfff org.ardupilot.apmplanner2 (*) <ef9ffd08-5c39-319e-a14f-cb9bb9436bd0> /Applications/apmplanner2.app/Contents/MacOS/apmplanner2
       0x1a6c56000 -        0x1a6ce4553 dyld (*) <2237410f-d39c-30ce-9a94-13aacb66b766> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
[... 114 lines cut ...]
      "rolloutId" : "63582c5f8a53461413999550",
      "factorPackIds" : {

      },
      "deploymentId" : 240000002
    },
    {
      "rolloutId" : "62c73fd17cce0d1b0bcb8a02",
      "factorPackIds" : {
        "SIRI_DIALOG_ASSETS" : "648719632dccf40610e755fa"
      },
      "deploymentId" : 240000134
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000013
    }
  ]
}
}

```

Arne-W, 2023-07-14:
@Iragne Could you please look??

Iragne, 2023-07-14:
I'm looking at it

Iragne, 2023-07-14:
@Arne-W and @valeriyvan  a pr is arriving correcting the issue associated to .gitignore 
@valeriyvan to make sure you will compile properly the app you need to do the following 
make distclean 
qmake -spec macx-clang apm_planner.pro
make -j8        
You will have a fresh new env and the framework will be added into the app.
if you don't do it you will keep to have the error at the start

Iragne, 2023-07-14:
here the PR https://github.com/ArduPilot/apm_planner/pull/1293

commit b7e7dc78 (2023-07-15): Build: Correction of the “.gitignore” and uploading missing file for SDL2 Fixes #1292
closed 2023-07-15 by Arne-W