ArduPilot/apm_planner #1301: AP says: AP: got MISSION_ITEM; GCS should send MISSION_ITEM_INT

AndKe, 2023-09-11:
reproduced in SITL, when uploading mission to ArduPlane, console says 
"AP: got MISSION_ITEM; GCS should send MISSION_ITEM_INT"

AndKe, 2023-10-20:
Also: AP2 is unable to request a mission plan from Ardupilot (also reproducible in SITL)
If a mission plan is requested by another GCS (like MAVProxy, AP2 will actually populate the plan just fine.

To reproduce:
Start SITL (make or load mission)
Connect AP2 
observe that there is no mission
In AP2 - observe that "read" and "refresh" buttons do nothing.
in SITL terminal , enter "wp list"  (requesting mission)
- observe that mission is properly populated.

I suspect it may be due to the use of https://mavlink.io/en/messages/common.html#MISSION_REQUEST
(note the red text)

@Arne-W  - are you familiar with this/what do you think?  (I see no obvious way to see MAVLINK messages used)

billbonney, 2025-07-01:
I was able to uploade and download missions with Plane 4.6.1. But need to investigate if this was fixed.