ArduPilot/apm_planner #1302: Graph: unable to plot BAT.EnrgTot

AndKe, 2023-09-15:
@Arne-W Please check why EnrgTot field won't plot (selecting it does nothing)
Suitable test-log https://drive.google.com/file/d/1mW0g0orfJ7wGICsTKXrKYz8kJ28TZhLB/view?usp=drive_link

[image]

- tested on current master, as well as older .appimage

Arne-W, 2023-09-15:
@AndKe I will have a look!

Arne-W, 2023-09-17:
@AndKe The problem is the dot in [W.h]. The data model separates the parts of a value row by using a dot like "BAT.Instance:0.CurrTot [Ah]".  When retrieving data we split the name at every dot resulting in "BAT" "Instance:0" "CurrTot [Ah]". Then there is a sanity check dropping everything which have more than 3 parts.
So splitting the name "BAT.Instance:0.EnrgTot [W.h]" at every dot results in "BAT" "Instance:0" "EnrgTot [W" "h]" which are 4 parts  and the sanity check drops the request making a logentry.
I will add some code removing dots within the SI unit.

commit 8ab9f961 (2023-10-27): Graphview: Fixed handling of unit data containing a dot like W.s. Dot will be removed. Fixes #1302
closed 2023-10-27 by Arne-W