ArduPilot/apm_planner #1306: Graph: The curves of the secondary menu are chaotic

datour, 2023-10-23:
i use two Batterys and two Battery-Monitor of DroneCAN on my copter，there use the same CAN-bus with CAN1.
when i checked the cpoter log , i find the curves of the secondary menu are chaotic，show it on this picture：
[image]

look at this，the curves is intersect when the secondary-menu data Out-of-sync （like BAT-Instance:0 and BAT-Instance:1 ，Index=3194796  in this picture），
[image]

it  will happen on every log data Out-of-sync:
[image]
[image]
[image]

The two batteries are separate devices，In theory, they will update their information at 1Hz ，But something unexpected may be happening，and I can't guarantee that their data is synchronized，maybe it have some  solv with graphically .

Arne-W, 2023-10-23:
@datour Hi thank you for the detailed description - which apm planner version are you using and which ardupilot version?

Arne-W, 2023-10-23:
Ok I looked at the source code and you are right for your use case it does not work. We assume that the order of an indexed type is always the same. Therefore we can access the data by using an iterator which takes every 2nd or every third field. 
In your case this assumption is simply wrong and the plotting fails.
Nevertheless it would be cool if you could upload the log somewhere so I can fetch it and reproduce this issue. Perhaps we can fix it.

Arne-W, 2023-11-12:
@datour I made a fix for your problem. If you can build the planner it would be nice if you could test it. If you can't build let me know - perhaps I can provide an appimage for testing.

commit df960ed4 (2023-11-17): GraphView: Fix the handling of indexed data with gaps Fixes #1306 #1307