ArduPilot/apm_planner #1307: Indexed types are processed incorrectly in the presence of gaps

mbuzdalov, 2023-11-05:
First of all, thank you for supporting indexed types!

However, the current code seems to think that, whenever an indexed type is present, the log entries will always come in strict order. I inferred this from the implementation of `LogdataStorage::getValues` and `break;  // if the index does not increase anymore we can stop` in `LogdataStorage::setupUnitData`, both in `src/ui/Loghandling/LogdataStorage.cpp`.

In my case, I have logging rate limiting enabled, so log entries coming normally at a very high rate (the ESC ones, for instance) are subsampled, and there is no longer any kind of monotonicity in terms of indices. What is more, even the actual sources of these log data can provide the data asynchronously (e.g. ESCs that talk over bidirectional DShot), so even without logging rate limiting the assumptions are wrong.

My first attempt to fix it this evening was unsuccessful, as the assumption of index monotonicity was taken really seriously in the codebase, and it seems that small changes cannot solve it. Could one please fix it, or advise me how to begin fixing?

Arne-W, 2023-11-10:
@mbuzdalov This is a duplicate of #1306.
Yes, you are right it is not a small change - the current attempt does not process any data before end of parsing and therefore did not know which data type is an idexed one and what is the max index. This was no problem as long as there were no gaps in the indexed data. But it seems that we have to adapt the parsing process to handle those data correctly.
I will try to fix it in the next 2 weeks I hope this is "fast" enough. It would be really cool if you could provide some logs with gaps for testing.

mbuzdalov, 2023-11-10:
[23-11-05_15-28-03.bin.zip](https://github.com/ArduPilot/apm_planner/files/13323767/23-11-05_15-28-03.bin.zip)

Here is an example. Check what ESC entries look like (e.g. ESC.RPM), there are four instances in fact, but only two are displayed, with the actual four instances being split in a funny way among these two. My new EDT entries also suffer from that.

Arne-W, 2023-11-12:
@mbuzdalov Thanks for the logfile. The fix was not as complicated as expected. I thought that we have to change data structure but it worked without that. What makes parsing somehow complicated is the fact that you can't assume that the description of a data type is complete before you get the first measurement of that type. So you have to parse all data, store it and at the end of parsing you have to do some sort of post processing to complete the datamodel...

It would be cool if you could test the pull request and check if it fixes the problem for you. I tested with your logfile and it worked.

mbuzdalov, 2023-11-12:
Your PR indeed fixes the problem for me, thank you very much!

Arne-W, 2023-11-17:
You are welcome :smile:

closed 2023-11-17 by Arne-W